PRBonn/vdbfusion #2: Extrinsics in RGBD integration

Willyzw, 2022-02-11:
Hallo, while trying out the example for RGB-D image based integration, I notice that the extrinsic/global poses are inverted. Which shall I interpret as `world to camera` transformation? Since normally the global poses are given as `camera to world`.

Lets take an example here, the global pose is inverted for scannet dataset. 
https://github.com/PRBonn/vdbfusion/blob/40ad3021c271c26d638672531818f86d2c91d3b8/examples/python/datasets/scannet.py#L33

Then the RGBD point cloud is transformed using the inverted pose here
https://github.com/PRBonn/vdbfusion/blob/40ad3021c271c26d638672531818f86d2c91d3b8/examples/python/datasets/scannet.py#L61-L63

So far is fine, as I roughly remember `open3d` indeed assumes an extrinsic from world to camera. However, apparently this inverted pose is further fed to the integration pipeline regarded as the origin. I am wondering whether this is really the purpose?
https://github.com/PRBonn/vdbfusion/blob/40ad3021c271c26d638672531818f86d2c91d3b8/examples/python/datasets/scannet.py#L74

Willyzw, 2022-02-11:
Thanks for the reply. Yes it solve my confusion. The interface expects the point clouds as well as the origins in global frame. Thus by contrast to `open3d`, which assumes an extrinsic pose from global to local frame, the dataloader needs to return normal `uninverted` pose.

closed 2022-02-11 by Willyzw

ShrutheeshIR, 2022-04-14:
Hey @Willyzw , I'm also trying to use an RGBD sensor instead of Lidar. Could you give a rough outline of steps you used. Thanks!