PRBonn/vdbfusion #6: How to get TSDF volume?

anastasiia-kornilova, 2022-02-21:
Dear authors, thank you for such awesome tool, it is working like a charm! I wonder, what is the fastest way to get TSDF volume as numpy array, as well its volume bounds?

nachovizzo, 2022-02-21:
I'm glad to hear you find the tool useful :) I actually have a `vdb_to_numpy` project which is not yet open-source. If you need it with some urgency please drop me an email and I will send the code privately.

In the meantime, your best friend is `OpenVDB`. I highly recommend building the project [from source](https://github.com/AcademySoftwareFoundation/openvdb) from the tip of master, since it has some important changes I had to make there.  Make sure yo use the following flags when building (taken from [here](https://github.com/PRBonn/vdbfusion/blob/126143faa105d39cffd14b370b18356e9e3d1980/3rdparty/OpenVDB/OpenVDB.cmake#L18)):
```sh
             -DCMAKE_POSITION_INDEPENDENT_CODE=ON
             -DOPENVDB_BUILD_PYTHON_MODULE=OFF
             -DOPENVDB_BUILD_VDB_PRINT=OFF
             -DOPENVDB_CORE_SHARED=OFF
             -DOPENVDB_CORE_STATIC=ON
             -DOPENVDB_CXX_STRICT=OFF
             -DUSE_CCACHE=ON
             -DUSE_STATIC_DEPENDENCIES=ON
             -DUSE_ZLIB=OFF
```

Once you have OpenVDB with the `pyopenvdb` bindings installed you **must** compile `VDBFusion` from the source. You can check the [INSTALL Readme](https://github.com/PRBonn/vdbfusion/blob/main/INSTALL.md) but is as simple as cloning this repo and doing `make install` on the root directory.

Once your dev environment is ready, the rest is quite easy and straight forward :) 

```python
import vdbfusion

vdb_volume = vdbfusion.VDBVolume(voxel_size,
                                 sdf_trunc,
                                 space_carving
dataset = Dataset(...)

for scan, origin in dataset:
    vdb_volume.integrate(scan, origin)
```

And then to play around with the TSDF volumes just:

```py
tsdf = vdb_volume.tsdf
print(tsdf.evalActiveVoxelBoundingBox())
```

All the documentation for the python interface is at the [OpenVDB website](https://www.openvdb.org/documentation/doxygen/python.html) Is not super complete, but a good starting point.

For more examples on how to use this within vdbfusion you can also check: https://github.com/PRBonn/vdbfusion/blob/126143faa105d39cffd14b370b18356e9e3d1980/examples/python/vdbfusion_pipeline.py#L86

Last but not least, if you are working with TSDF volumes you can also check my comments on [This PR](https://github.com/PRBonn/vdbfusion/pull/4#issuecomment-1036227859). I will add propper documentation to the project whenever I find the time

Let me know if this helps

anastasiia-kornilova, 2022-02-21:
Wow, thanks for fast response! I suppose all steps are clear for me. I will try and give you feedback.

closed 2022-03-18 by nachovizzo