PRBonn/vdbfusion #7: ROS Integration example

alex-ssom, 2022-03-17:
Hi,

I'm currently looking at integrating this algorithm to our ROS based robot software stack. Our robot is equipped with a LiDAR, a RealSense D435i and encoders. We're currently using the gmapping SLAM algorithm to provide the transform from the `/map` frame to the `/realsense_depth_optical frame`.

I was wondering if you managed to integrate this library with ROS in the past and if you have pointers on how to integrate it with our robot?

Thanks!
Alex

nachovizzo, 2022-03-18:
Hello,

Thanks for opening this issue. We are currently working on a `vdbfusion_ros` package, which will fully support this library in ROS. Please stay tuned, I will keep you posted on this issue and let you know whenever we open-source the ROS API

Best regards,

skohlbr, 2022-04-07:
Hi @nachovizzo can you provide an (rough) indication of when you'd make the `vdbfusion_ros` available? Would like to avoid duplicating work on this.

nachovizzo, 2022-04-11:
Hi, sorry for the late reply. I guess that by the end of this month we will have something ready to open-source, I will keep you posted on this issue

nachovizzo, 2022-06-09:
Dear @skohlbr , sorry for the late reply. A ROS wrapper is already available at https://github.com/PRBonn/vdbfusion_ros