PRBonn/vdbfusion #9: How about using Open3D to do the TSDF?

qpc001, 2022-03-21:
Open3D supported GPU.

nachovizzo, 2022-03-21:
Open3D has recently added support for GPU-TSDF. Besides the fact that this effort was done almost in parallel with VDBFusion, the VDBFusion API is a bit cleaner to use, since it only requires point clouds as the input, while Open3D requires an image (or range-image for LiDAR). On the other hand, VDBFusion make use of state-of-the-art sparse voxel data structures (VDB) while Open3D made their own, pretty well engineered I must say, based on the classical voxel-hashing approach by Nießer et al. More details on how the new reconstruction systems work in Open3D can be found at [this paper](https://arxiv.org/pdf/2112.02779.pdf). 

Basically, they solve the same problem but with 2 different approaches. I hope this replies to your question

closed 2022-03-21 by nachovizzo

mpottinger, 2022-10-28:
> Open3D has recently added support for GPU-TSDF. Besides the fact that this effort was done almost in parallel with VDBFusion, the VDBFusion API is a bit cleaner to use, since it only requires point [...]

I am thinking some kind of hybrid approach may work. Implement voxel hashing on the GPU and then periodically stream out of GPU and merge into a vdb volume. Not sure if it's practical or actually makes sense but I will give it a try.