PRBonn/vdbfusion #21: cannot make install succeed

yinloonga, 2022-06-27:
hello, I want to use C++ API, then I followed your instructions **Build from source in Linux**. But, when I tried to use `sudo make instal`, I got some messages.
[make_install.log](https://github.com/PRBonn/vdbfusion/files/8990965/make_install.log)  

You know, most of the `CMakeLists.txt` projects can be installed as the below instructions:
```
git clone https://github.com/PRBonn/vdbfusion.git
cd vbdfusion && mkdir build && cmake ..
make -j4
```
Some things looked better than before( Because I can see the download packages process). However, I also got some messages; 
[make_j4.log](https://github.com/PRBonn/vdbfusion/files/8991486/make_j4.log)

As the terminal output to screen, I tried to find what happened in `make`.
[bootstrap.log](https://github.com/PRBonn/vdbfusion/files/8991508/bootstrap.log)

It just looks like having some problems with `boost`, I found some answers to this question([undefined reference to](https://github.com/boostorg/build/issues/468)), but I don't know if it's correct. By the way, my system(Ubuntu 20.04) has installed **boost(1.71)**, and your packages also need to reinstall boost as 3third libraries(the required boose version also is 1.71). 

I don't know the next what I should do. Please give some tips.
So, I wish you can help me solve this problem when you have little time. Thank you!

nachovizzo, 2022-06-27:
So let me see if I understand.

If you want to install **JUST** the C++ API, then it should be straight forward. Just follow the instructions in the [Install.md](https://github.com/PRBonn/vdbfusion/blob/main/INSTALL.md#installing-the-c-api). As mentioned there, you need to make sure that all the 3rdparty dependencies are installed locally. This means you should install the OpenVDB library and also Eigen3.

Now about your errors.

## `make_install.log`

This looks like you are trying to build the pip package, so don't run `make` or `pip install .` if you don't need them

### `make_j4.log`

From inspecting the log I can tell you didn't install the dependencies as requested, and it's failing (although the error is somehow strange, it says `Consult 'bootstrap.log' for more details`, which indeed exhibits a linking error I haven't seen before. Anyways, you must install the dependencies from source in case you need the C++ API. Since to make sure everything is working system-wise, you must have those installed beforehand. Luckilly is not a big challenge, just follow the instructions on the main [Dockerfile](https://github.com/PRBonn/vdbfusion/blob/main/docker/builder/Dockerfile) and that should be it

```sh
# Install C++ Dependencies
sudo apt-get update && sudo apt-get install --no-install-recommends -y \
    libblosc-dev \
    libboost-iostreams-dev \
    libboost-numpy-dev \
    libboost-python-dev \
    libboost-system-dev \
    libeigen3-dev \
    libtbb-dev

# Install OpenVDB from source
git clone --depth 1 https://github.com/nachovizzo/openvdb.git -b nacho/vdbfusion \
    && cd openvdb \
    && mkdir build && cd build \
    && cmake \
    -DCMAKE_POSITION_INDEPENDENT_CODE=ON \
    -DUSE_ZLIB=OFF \
    ..\
    && sudo make -j$(nproc) all install \
```

## Summary

To achieve your goal, just install the dependencies as specified in the last step and try to `make install` again

Please close this issue if this fix your problem, and if you feel like the Installation instructions are not clear, feel free to submit a merge request with updated instructions.

yinloonga, 2022-06-28:
Thank you very much for your prompt reply. And I'm sorry for my late reply. I just followed your instructions. Then I got messages in the terminal:

```
Scanning dependencies of target vdbfusion
[ 33%] Building CXX object src/vdbfusion/vdbfusion/CMakeFiles/vdbfusion.dir/MarchingCubes.cpp.o
[ 66%] Building CXX object src/vdbfusion/vdbfusion/CMakeFiles/vdbfusion.dir/VDBVolume.cpp.o
[100%] Linking CXX static library libvdbfusion.a
[100%] Built target vdbfusion
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/libvdbfusion.a
-- Installing: /usr/local/include/vdbfusion/VDBVolume.h
-- Installing: /usr/local/lib/cmake/VDBFusion/VDBFusionTargets.cmake
-- Installing: /usr/local/lib/cmake/VDBFusion/VDBFusionTargets-release.cmake
-- Installing: /usr/local/lib/cmake/VDBFusion/VDBFusionConfig.cmake
-- Installing: /usr/local/lib/cmake/VDBFusion/VDBFusionConfigVersion.cmake
```

It looks like the installation was successful。 Thank you so much.

---
You know, I want to learn about **VDBFusion C++ API**. Then I compiled the **examples-cpp** folder. It's used **KITTI 07** dataset. 
I read your paper about **5.Experiments**, you also test on the RGB-D dataset **Cow and Lay dataset**. I have seen other paper also used **Cow and Lay dataset**, and my part work also about RGB-D cameras. So, can you provide the previous test program code for cow_lady_dataset? Than you.

nachovizzo, 2022-06-28:
I have a hideous C++ implementation for the cow and lady dataset, which is going to confuse you instead of help. You have 2 options then:

1. Use the [vdbfusion_ros](https://github.com/PRBonn/vdbfusion_ros) we released a few weeks ago, where there is an example with cow and lady
2. Write a C++ dataloader yourself, and make a contribution for this repository. I'd be more than happy to integrate it :)

yinloonga, 2022-06-28:
Ok, thank you! I'll try my best to try to make any improvements to this project in my future studies. 

However, I tried using **vdbfusion_ros** as you instructed and got some **Extract Triangle Mesh** that looked ok. But here, for `cow_and_lady` dataset, I don't know if it has something to do with the dataset itself when I try to get data on the topic `/camera/depth_registered/points` in `Rviz`, it gives an error. Here is a [video](https://user-images.githubusercontent.com/72967324/176181823-bf79946e-5bc2-4eef-be12-ab8a4602623d.mp4) showing what happened.
Obviously, I have forced the setting The `world` is the **fixed base link**. I don't know why this is happening, can you see what the problem is, thanks. 

Really appreciate your previous reply. Thanks again.

nachovizzo, 2022-07-04:
This sounds like a ROS problem, feel free to open a new issue on the vdbfusion-ros in case you are still struggling.

Best wishes!

closed 2022-07-04 by nachovizzo