PRBonn/vdbfusion #24: Worse results than the paper displayed

Rotatingpencil, 2022-07-22:
I'm running the kitti 00 sequence with python API, but I got the result below:
[image]
[image]

which is much worse than the paper displayed:

[image]

I got this result without editing the config file.

And I repeated it by changing the ```fill_holes``` to True, but the result did not change.

Why there are two different results?

Rotatingpencil, 2022-07-22:
For example, there are overlaps of cars and bumpy roads.

nachovizzo, 2022-07-22:
Hello @Rotatingpencil, that looks like pose estimation errors. Usually the so-called "ground truth" poses from KITTI are not enough. You can give it a try with these poses:
[00.txt](https://github.com/PRBonn/vdbfusion/files/9167546/00.txt)

Just override the `00.txt` on your filesystem. Please let me know if this helps with the problem

Rotatingpencil, 2022-07-22:
Now the result seems perfect! 

How can I get the correct poses that you sent me, the poses I used before were downloaded from the official website, but it's wrong....

nachovizzo, 2022-08-08:
You can use any SLAM system, for example SuMa ;)

closed 2022-08-08 by nachovizzo