PRBonn/vdbfusion #27: Ubuntu 18 support or reimplementation

maulanaazhari, 2022-11-01:
Hi, really great work on the volumetric mapping! I really appreciate it!

I want to ask if there is a possibility of Ubuntu 18 support? If not, maybe can you tell me a pathway for reimplementing this framework on ubuntu 18.

I saw that the newer versions of openVDB don't support ubuntu 18 out of the box, so I want to use an older version of it for ubuntu 18, maybe version 5 or 6. Is there any feature that is only present in the newer version for this **vdbfusion**?

Thanks~

nachovizzo, 2022-11-02:
Hello @maulanaazhari , may I ask. Have you tried building the project from source? I guess it should kind of work. Just follow the instructions on the [install instructions](https://github.com/PRBonn/vdbfusion/blob/1d2547526a3c2c088ec12fe614fc9902c154b6a3/INSTALL.md) file. 

Just give it a try and let me know here ;) If it doesn't work, docker is always your friend in these cases and can help you

maulanaazhari, 2022-11-02:
Thanks for your reply @nachovizzo, I've tried to build it from source but I guess it has too many requirements that ubuntu 18 doesn't meet, like the compiler (GCC) and CMake version, even after changing those I still got errors.

and for me, Docker is not really an option.

Luckily I've found a workaround by using an older version of OpenVDB (ver 5) to work on ubuntu 18, but I only reimplement the C++ codes without the python3 binding and mesh reconstruction. So far it works quite well in ROS Melodic.

Thanks for the codes and insights!

nachovizzo, 2022-11-02:
Ok! I'm happy it worked out at the end

closed 2022-11-02 by nachovizzo