PRBonn/vdbfusion #28: more detail for dataset format

waveleaf27, 2022-12-01:
Hi, can you provide script to generate nuscene dataset or detailed format infos of data format,  We haven't been able to successfully run the sample code yet.  Thanks @niosus @nachovizzo

https://github.com/PRBonn/vdbfusion/blob/0704860c7c9dc0c2899faecd79047cb8c5340bb7/examples/python/datasets/nuscenes.py#L48

niosus, 2022-12-03:
Please be mindful who you tag in the questions @waveleaf27. This project is developed by @nachovizzo and I (unfortunately 😉) have nothing to do with it.

waveleaf27, 2022-12-04:
Sorry for the wrong tag @niosus

nachovizzo, 2022-12-12:
@waveleaf27 Sorry for the late reply. I just updated the nuScenes dataloader making it completly indpendent from KITTI. You can use it and let me know if it's now working for you! Thanks :)

This is how my directory looks like after downloading from the official website
```sh
~/data/nuScenes tree -L 2 .
.
├── maps
│   ├── 36092f0b03a857c6a3403e25b4b7aab3.png
│   ├── 37819e65e09e5547b8a3ceaefba56bb2.png
│   ├── 53992ee3023e5494b90c316c183be829.png
│   └── 93406b464a165eaba6d9de76ca09f5da.png
├── samples
│   ├── CAM_BACK
│   ├── CAM_BACK_LEFT
│   ├── CAM_BACK_RIGHT
│   ├── CAM_FRONT
│   ├── CAM_FRONT_LEFT
│   ├── CAM_FRONT_RIGHT
│   ├── LIDAR_TOP
│   ├── RADAR_BACK_LEFT
│   ├── RADAR_BACK_RIGHT
│   ├── RADAR_FRONT
│   ├── RADAR_FRONT_LEFT
│   └── RADAR_FRONT_RIGHT
├── sweeps
│   ├── CAM_BACK
│   ├── CAM_BACK_LEFT
│   ├── CAM_BACK_RIGHT
│   ├── CAM_FRONT
│   ├── CAM_FRONT_LEFT
│   ├── CAM_FRONT_RIGHT
│   ├── LIDAR_TOP
│   ├── RADAR_BACK_LEFT
│   ├── RADAR_BACK_RIGHT
│   ├── RADAR_FRONT
│   ├── RADAR_FRONT_LEFT
│   └── RADAR_FRONT_RIGHT
├── v1.0-mini
│   ├── attribute.json
│   ├── calibrated_sensor.json
│   ├── category.json
│   ├── ego_pose.json
│   ├── instance.json
│   ├── log.json
│   ├── map.json
│   ├── sample_annotation.json
│   ├── sample_data.json
│   ├── sample.json
│   ├── scene.json
│   ├── sensor.json
│   └── visibility.json
└── v1.0-trainval
    ├── attribute.json
    ├── calibrated_sensor.json
    ├── category.json
    ├── ego_pose.json
    ├── instance.json
    ├── log.json
    ├── map.json
    ├── sample_annotation.json
    ├── sample_data.json
    ├── sample.json
    ├── scene.json
    ├── sensor.json
    └── visibility.json
```

commit 88f94a1e (2022-12-12): Update nuScenes dataloader, fix #28
closed 2022-12-12 by nachovizzo