PRBonn/vdbfusion #30: sporadic free(): invalid pointer Aborted (core dumped) when creating VDBVolume

joerowelll, 2023-01-16:
I am getting the (sporadic but persistent) error:
```
free(): invalid pointer
Aborted (core dumped)
```
when creating a VDBVolume, using 
```
vdb_volume = vdbfusion.VDBVolume(voxel_size, sdf_trunc)
```
From what I can tell from debugging, the volume is created in VDBVolume.cpp  in the code block below, and this code runs fine, however nothing else is called and the error still occurs afterwards.
```
VDBVolume::VDBVolume(float voxel_size, float sdf_trunc, bool space_carving /* = false*/)
    : voxel_size_(voxel_size), sdf_trunc_(sdf_trunc), space_carving_(space_carving) {
    std::cout << "VDBVolume constructor" << std::endl;
    tsdf_ = openvdb::FloatGrid::create(sdf_trunc_);
    std::cout << "VDBVolume constructor: Float Grid Created" << std::endl;
    tsdf_->setName("D(x): signed distance grid");
    tsdf_->setTransform(openvdb::math::Transform::createLinearTransform(voxel_size_));
    tsdf_->setGridClass(openvdb::GRID_LEVEL_SET);
    std::cout << "VDBVolume constructor: Set TSDF properties" << std::endl;
    weights_ = openvdb::FloatGrid::create(0.0f);
    weights_->setName("W(x): weights grid");
    weights_->setTransform(openvdb::math::Transform::createLinearTransform(voxel_size_));
    weights_->setGridClass(openvdb::GRID_UNKNOWN);
    std::cout << "VDBVolume constructor: Set weights" << std::endl;
}
```

To recreate:
Build vdbfusion from source as per README.md.
Run Python script
```
import vdbfusion
vdbfusion.VDBVolume(1,1)

```

nachovizzo, 2023-01-16:
Could you give more details on the system, datasets, etc?

joerowelll, 2023-01-16:
I am running Ubuntu 20.04, and building something for my own dataset of scans and poses. Although the error occurs at instantiation of the VDBVolume, before even populating it with the dataset.

nachovizzo, 2023-01-17:
Could you check the version of OpenVDB Installed?

joerowelll, 2023-01-17:
I am using the nachovizzo/openvdb , version 9.1.1.

joerowelll, 2023-01-18:
Solved it, it seems CMake was linking to a system version of TBB , rather than the mambaforge version. Changing this and re-building openvdb fixed the issue.

closed 2023-01-18 by joerowelll