PRBonn/vdbfusion #34: ERROR: No metrics available, please compile with PYOPENVDB_SUPPORT

zang09, 2023-02-23:
Thank you for sharing your great work!
VDBFusion is work well, but i have some problem..

I'm having a hard time building vdbfusion with `pyopenvdb`.
I just follow [this](https://github.com/PRBonn/vdbfusion/blob/main/INSTALL.md#build-from-source-in-a-conda-environment) instructions on my M2 mac with conda.
And get this message : No metrics available, please compile with PYOPENVDB_SUPPORT.

Is there any idea?

nachovizzo, 2023-03-24:
Hello there, sorry for the late reply!

Did you manage to build OpenVDB from source first? This would be the most tricky part on a M2 platform

nachovizzo, 2023-03-24:
I see, would you mind re-opening this issue in OpenVDB, since it's not the library from this particular repo. Can you also sahre the error msg? not from vdbfusion, but from OpenVDB. Just open a python shell inside the conda env and do `import pyopenvdb`.

closed 2023-03-24 by zang09

HerculesJL, 2023-08-16:
> Thank you for sharing your great work! VDBFusion is work well, but i have some problem..
>
> I'm having a hard time building vdbfusion with `pyopenvdb`. I just follow [...]

Hello! Can you tell me how to troubleshoot this error? Thank you very much

nachovizzo, 2023-08-16:
@HerculesJL I'm currently working on migrating the build system, also for m2. So stay tuned, soon should be fixed