PRBonn/vdbfusion #37: Compile with PYOPENVDB_SUPPORT_ENABLED

LukasDeGreve-FM, 2023-09-21:
Hello

I recently came across this library and it is doing almost all the things I need for a project I am working on. Thank you! I want to experiment with using different weights, but unfortunately when using the python API I encounter the following error:
`NotImplementedError: Please compile with PYOPENVDB_SUPPORT_ENABLED`

I called the following function, similar to the example in the paper
`vdb_volume.integrate(pcd_numpy, np.identity(4), lambda sdf: 1 if sdf < eps else np.exp(-sigma * (sdf-eps) ** 2))`
and simply installed vdbfusion using pip. The python version is 3.10

I have very little experience with compiling libraries from source and so was wondering what should be done exactly to fix this issue. Do I simple follow the steps from [the install instructions](https://github.com/PRBonn/vdbfusion/blob/main/INSTALL.md#build-from-source-in-linux)?

Thanks in advance

saurabh1002, 2024-04-11:
Hello @LukasDeGreve-FM,

Yes you will have to build from source to overcome that error.

There are two ways to build from source with OpenVDB Python support.
1. If you are using OpenVDB installed on your local machine, which is the default setting [here](https://github.com/PRBonn/vdbfusion/blob/c67735182de0fcadb2ad8f32afdf8a1ca61f0cef/CMakeLists.txt#L29) when building from source,  simply follow the instructions  [here](https://github.com/PRBonn/vdbfusion/blob/main/INSTALL.md) under *Building OpenVDB from source* to install OpenVDB locally.
2. Alternatively, you can set [this flag](https://github.com/PRBonn/vdbfusion/blob/c67735182de0fcadb2ad8f32afdf8a1ca61f0cef/CMakeLists.txt#L29) to `OFF` and let the build system install OpenVDB for you. In this case you need to manually edit [this](https://github.com/PRBonn/vdbfusion/blob/c67735182de0fcadb2ad8f32afdf8a1ca61f0cef/3rdparty/OpenVDB/OpenVDB.cmake#L45) option to `ON`, to instruct the build system to build OpenVDB with python support.

Either of these steps, along with the ones already explained in the `INSTALL.md` file should fix it hopefully :)

LukasDeGreve-FM, 2024-04-11:
Hey @saurabh1002 

Thanks for the response. I was eventually able to solve this problem on my own, but forgot to update this issue. Sorry about that.

Besides doing what you suggested, I also encountered a problem with recognizing that pyopenvdb was installed. Namely, in [lines 29-32 of this script](https://github.com/PRBonn/vdbfusion/blob/main/src/vdbfusion/pybind/CMakeLists.txt), the value of `PYOPENVDB_SUPPORT` was always false, despite having pyopenvdb installed. The source of this problem was likely the way my python environments and virtual environments are set up. I forced it to compile the additional bindings and everything worked in the end

saurabh1002, 2024-04-11:
Great to know. I will close this issue in that case.

closed 2024-04-11 by saurabh1002

l00p3, 2024-04-12:
@LukasDeGreve-FM Could you please share what exactly do you mean by `I forced it to compile the additional bindings`? It would be helpful if you can share what u did exactly.

LukasDeGreve-FM, 2024-04-12:
@l00p3 
In the above mentioned script (pybind/CMakeLists.txt), I removed lines 28-32, because the value of PYOPENVDB_SUPPORT would always be false on my system. Then I also remove the if statement on lines 33 and 38, meaning that the code on lines 33-37 would always execute

Hope this helps