PRBonn/vdbfusion #42: Question about Eye-to-hand Scenario Usage

zyaqcl, 2024-04-23:
Hi

I have a depth camera mounted on the base and an object mounted on the end effector of robot arm. After scanning, _hand_T_cam_ was calculated and used as the pose and sent along the scan result for fusion. I thought I could get the result including  a clear model of the object and a messy background. However, on the contrary, I got a clear background and a messy foreground. Why is that?
[image]

Thanks

benemer, 2024-04-29:
Note that the points that are passed to the `integrate` function need to be in a common frame. Is that the case on your side?

zyaqcl, 2024-04-30:
@benemer Thanks for your reply. I have all the point clouds mapped to the same **Base** frame and performed point cloud alignment and got transformation for each point cloud. What should I choose as the input of `intergrate`? Should that be aligned point clouds with identical poses?

benemer, 2024-04-30:
These should be the transformed points and the poses in the same coordinate frame.

So, if you have transformations from each local frame to the base frame, pass those poses together with the points transformed to the base frame.

zyaqcl, 2024-04-30:
@benemer Thank you fro your patience and apologies for not being specific.
I have a set of point clouds under base frame marked as <sup>Base</sup>**P**, which are transfromed from their own local frame with <sup>Base</sup>**T**<sub>Local</sub>.
After performing point cloud alignment, I got <sup>Align</sup>**T** for each point cloud. With <sup>Base</sup>**P**. `transform`(<sup>Align</sup>**T**) I can get <sup>Aligned</sup>**P**.
I have tried 4 setups for the input.
If input is (<sup>Base</sup>**P**, <sup>Base</sup>**T**<sub>Local</sub>) I can get correct but not aligned points.
Input (<sup>Base</sup>**P**, <sup>Align</sup>**T**@<sup>Base</sup>**T**<sub>Local</sub>)  I get similar result as upper one.
If input is (<sup>Aligned</sup>**P**. <sup>Align</sup>**T**@<sup>Base</sup>**T**<sub>Local</sub>) or (<sup>Aligned</sup>**P**. <sup>Base</sup>**T**<sub>Local</sub>) , the results do not seem correcrt.

zyaqcl, 2024-05-07:
@benemer 
Thank you. Your solution seems correct.
The base frame is the frame of robot arm end effector which the object is mounted on.
I did the alignment with `open3d.registration.registration_icp`. The <sup>Align</sup>**T** indicates the result.
Previously I used the <sup>Align</sup>**T** incorrectly so the results are wrong.
It seems that there is little diference between using (<sup>Aligned</sup>**P**, <sup>Base</sup>**T**<sub>Local</sub>) and (<sup>Aligned</sup>**P**, <sup>Align</sup>**T** @ <sup>Base</sup>**T**<sub>Local</sub>)

closed 2024-05-07 by benemer