PRBonn/vdbfusion #51: Error when runing kitti pipeline

ignotuspeverel, 2025-02-26:
How to fix this ?

Traceback (most recent call last):
  File "/home/ignotus/code/vdbfusion/examples/python/kitti_pipeline.py", line 27, in <module>
    argh.dispatch_command(main)
  File "/home/ignotus/mc3/envs/vf/lib/python3.9/site-packages/argh/dispatching.py", line 472, in dispatch_command
    dispatch(parser, *args, **kwargs)
  File "/home/ignotus/mc3/envs/vf/lib/python3.9/site-packages/argh/dispatching.py", line 199, in dispatch
    return run_endpoint_function(
  File "/home/ignotus/mc3/envs/vf/lib/python3.9/site-packages/argh/dispatching.py", line 270, in run_endpoint_function
    return _process_command_output(lines, output_file, raw_output, always_flush)
  File "/home/ignotus/mc3/envs/vf/lib/python3.9/site-packages/argh/dispatching.py", line 290, in _process_command_output
    for line in lines:
  File "/home/ignotus/mc3/envs/vf/lib/python3.9/site-packages/argh/dispatching.py", line 415, in _execute_command
    for line in result:
  File "/home/ignotus/mc3/envs/vf/lib/python3.9/site-packages/argh/dispatching.py", line 395, in _call
    result = function(*positional_values, **values_by_name)
  File "/home/ignotus/code/vdbfusion/examples/python/kitti_pipeline.py", line 22, in main
    pipeline.run()
  File "/home/ignotus/code/vdbfusion/examples/python/vdbfusion_pipeline.py", line 54, in run
    self._run_tsdf_pipeline()
  File "/home/ignotus/code/vdbfusion/examples/python/vdbfusion_pipeline.py", line 70, in _run_tsdf_pipeline
    scan, pose = self._dataset[idx]
  File "/home/ignotus/code/vdbfusion/examples/python/datasets/kitti_odometry.py", line 60, in __getitem__
    return self.scans(idx), self.poses[idx]
  File "/home/ignotus/code/vdbfusion/examples/python/datasets/kitti_odometry.py", line 66, in scans
    return self.read_point_cloud(idx, self.scan_files[idx], self.config)
  File "/home/ignotus/code/vdbfusion/examples/python/utils/cache.py", line 54, in wrapper
    cls.cache.set(key, result, expire, tag=tag, retry=True)
  File "/home/ignotus/mc3/envs/vf/lib/python3.9/site-packages/diskcache/core.py", line 808, in set
    self._row_insert(db_key, raw, now, columns)
  File "/home/ignotus/mc3/envs/vf/lib/python3.9/site-packages/diskcache/core.py", line 857, in _row_insert
    sql(
sqlite3.OperationalError: no such column: "size" - should this be a string literal in single-quotes?

saurabh1002, 2025-05-26:
@ignotuspeverel Could you provide the exact command that you ran leading to this error?

ignotuspeverel, 2025-05-26:
First I try to use the pip command to install VDBFusion but failed says no longer provided, there's other opening issue mentioning this. Then I try to use the building wheel/packages on the pypi with the earlier version to install the package which succeeded, but ran into this issue when runing the kitti example pipeline.

saurabh1002, 2025-05-26:
What OS are you trying to install on?